voneiden/cq-cam #1: Suggestion: clipping faces with OCCT

adam-urbanczyk, 2022-02-14:
You could consider clipping faces using OCCT instead of clipper (and thus operate on a more accurate representation). Clipping wires does not work, but planar faces do work (see `cq.Sketch` for inspiration). You'll be likely only dealing with closed wire so there is a 1-1 correspondence between faces and wires.

voneiden, 2022-02-14:
Thanks for the tip. I imagine the primary benefit here would be preserving circular/arc features and their later conversion to G2/G3 motion commands. However clipping open wires is still necessary for creating toolpath segments, for example here's my yesterday's attempt at generating a zigzag pocket:

[image]

I've also thought about recovering circular motion in post by detecting sequences of three or more points that match the expected footprint of an interpolated arc/circle.

adam-urbanczyk, 2022-02-15:
I think that this use case is also supported - you can e.g. cut edges with faces or edges with edges (though in the latter case you effectively split them).

Anyways, enough of my armchair advice. Cool project!

closed 2022-02-15 by adam-urbanczyk

voneiden, 2022-02-15:
Thanks Adam, I'll definitely give that a look at some point. I'm all for prioritizing code simplicity and minimizing dependencies with this project as it has some potential to become a long term maintenance project.