voneiden/cq-cam #16: Invalid G3 command in unstable branch

voneiden, 2022-10-26:
jmwright reported following gcode was generated

```
G0X35.8Y0Z3
F3G1Z0
G3I-35.8
G0Z3
```

I'll hazard a guess that the current implementation incorrectly optimizes away  X & Y coordinates for a full circle G2/G3 command. However as far as I remember it is handled correctly in in `fluent-experiments` branch.

jmwright, 2022-10-26:
Some background: [This question](https://openbuilds.com/threads/g2-g3-error-26.8717/) on the OpenBuilds forum has some discussion on error 26, which is the error that sparked this discussion. I was using [Candle CNC](https://github.com/Denvi/Candle) when I got the error, but it's a grbl error and not specific to Candle. Here is a [grbl error reference](https://resources.sienci.com/view/gs-grbl-alarm-error-codes/) that has error 26 listed and it's cause, which I think makes sense given your diagnosis.

voneiden, 2022-10-26:
Yep, thanks. [NIST RS274NGC Interpreter - Version 3](https://tsapps.nist.gov/publication/get_pdf.cfm?pub_id=823374) reads that in "Center Format Arc" (IJK) in XY-plane it is an error if "X and Y are both omitted". As it's a full circle, the correct command would be `G3I-35.8X35.8Y0`. Interestingly enough it does render correctly for me with ncviewer (just have to zoom out quite a bit). 

Looks like XYZ is completely missing in `unstable`: https://github.com/voneiden/cq-cam/blob/unstable/cq_cam/commands/base_command.py#L219

Branch `fluent-experiments` does have XYZ: https://github.com/voneiden/cq-cam/blob/fluent-experiments/cq_cam/command.py#L223

So this will be fixed once I manage to finalize the work on `fluent-experiments`.

jmwright, 2023-06-06:
Thanks @voneiden !