voneiden/cq-cam #23: visualize_job does not work unless you use cq_editor

giannissc, 2023-04-24:
I have recently switched from cq_editor to ocp_cad_viewer extension for vs code. Running visualize_job does not work but visualize_job_as_edges does

voneiden, 2023-04-24:
It may very well be that AIS based rendering requires changes to ocp_cad_viewer like it required for cq-editor ( https://github.com/CadQuery/CQ-editor/pull/317 ). For cq-editor these changes were pretty simple.

voneiden, 2023-04-29:
Preliminary testing shows that render performance is actually slightly better with 100k random lines using `TopoDS_Edge` rather than `AIS_Line`.

It is slightly slower to generate Edges though. 

However there doesn't seem to be any easy way of configuring the color of an edge in OpenCASCADE

voneiden, 2023-05-06:
Fixed by #31 - opened #36 for making the visualization more informative in ocp-vscode.

closed 2023-05-06 by voneiden