voneiden/cq-cam #38: Refactorings

giannissc, 2023-05-08:
- [ ] Optional[foo] vs Union[foo, None]
- [ ] Tuple[] -> tuple[]
- [ ] List[] -> list[]
- [ ] Dict[] -> dict[]
- [ ] `start` is not necessary for `Linear` commands (it can be read from `previous_command` ) but it is for `Circular` as it changes it gcode output
- [ ] `to_gcode()` overhaul
- [ ] `Drill` and `Surface3D` are classes but `profile` and `pocket` are method. We should be consitent

giannissc, 2023-05-08:
This are just a couple of inconsistencies I noticed and wanted to tackle next. Feel free to add your own to track the changes that are needed to be made. Also can you please clarify which style is the preferred in the first and last points? @voneiden

voneiden, 2023-05-08:
First) PEP 484 reads merely "As a shorthand for `Union[T1, None]` you can write `Optional[T1]`", so there's no official preference for either. SonarCloud does complain about `Union[T1, None]` though suggesting to replace it with union type expression `T1 | None` . I've not used these much, but seems also like a fairly reasonable option as it's shorter to type and doesn't require imports.

Not sure which one you refer to the last one as there have been a few edits. 

> There are a couple of NumPy deprecation warning when running pytestdue to typing. Should we do anything about them?

Is that covered by 5d8aec3 ? 

> Drill and Surface3D are classes but profile and pocket are method. Should we be consistent?

Yes, the class based is older structure and needs refactoring. Both were quite hastily bolted on the fluent api.  

>  Should i change update_tool to use shallow copies for self = self._add_operation("Tool Change", commands)?

The self override in the middle of the method is certainly a bit unconventional and would be neat if there's some other way to handle this.

> Some methods use typing information and some don't. Should we add to the rest as well?

Not a high priority, but at least I like having them.

voneiden, 2023-05-08:
For priority, the to_gcode overhaul could be pretty high on the list. One large inconsistency that exists is the precision (gcode_precision). There are multiple places all over that require rounding (gcode output, spline approximation, clipper offsetting) of values, and currently it's all hardcoded with placeholder values. Would be nice to get that sorted, but it's probably best to finalize it after the to_gcode overhaul is done.

giannissc, 2023-05-09:
> Is that covered by [5d8aec3](https://github.com/voneiden/cq-cam/commit/5d8aec3c592ece18c8931df68f340665d8b50f73) ?

I think so. I will keep it on the list to check later as well. I will break this into separate issues for different groups