voneiden/cq-cam #59: [BUG] Toolpath generation and Visualisation

giannissc, 2023-05-17:
I just checked the state of the code for the changes that I did so far and I noticed a couple of bugs bugs.

Edge visualization is broken but the resulting G-code is mostly correct

[image]
[image]

Toolpath generation introduced an additional G1 command that produces a diagonal movement that should be there
Toolpath generation introduces an extra movement before moving between adjacent wires adding unnecessary work

giannissc, 2023-05-17:
Correct Gcode:

```
(Job - Feedrate: 300 - Unit: Unit.METRIC)
G90 G54 G64 G50 G17 G94
G49 G40 G80
G21
G30
M3
(Job - Pocket)
G0 Z75
G0 X82.85 Y35.35 Z75
G0 X82.85 Y35.35 Z5
G1 X82.85 Y35.35 Z-0.25 F15
G1 X62 Y35.3 Z-0.25 F300
G1 X62 Y33.8 Z-0.25 F300
G1 X82.85 Y33.75 Z-0.25 F300
G1 X82.85 Y35.35 Z-0.25 F300
G0 X82.85 Y35.35 Z75
G0 X84.05 Y36.55 Z75
G0 X84.05 Y36.55 Z5
G1 X84.05 Y36.55 Z-0.25 F15
G1 X34.88 Y36.55 Z-0.25 F300
G1 X36.453 Y35.3 Z-0.25 F300
G1 X60.8 Y35.3 Z-0.25 F300
G1 X60.8 Y33.8 Z-0.25 F300
G1 X36.453 Y33.8 Z-0.25 F300
G1 X36.322 Y33.53 Z-0.25 F300
G1 X34.63 Y32.55 Z-0.25 F300
G1 X84.05 Y32.55 Z-0.25 F300
G1 X84.05 Y36.55 Z-0.25 F300
G0 X84.05 Y36.55 Z75
G0 X85.25 Y37.75 Z75
G0 X85.25 Y37.75 Z5
G1 X85.25 Y37.75 Z-0.25 F15
G1 X31.35 Y37.75 Z-0.25 F300
G1 X31.35 Y31.35 Z-0.25 F300
G1 X85.25 Y31.35 Z-0.25 F300
G1 X85.25 Y37.75 Z-0.25 F300
G0 X85.25 Y37.75 Z75
G0 X86.45 Y38.95 Z75
G0 X86.45 Y38.95 Z5
G1 X86.45 Y38.95 Z-0.25 F15
G1 X30.15 Y38.95 Z-0.25 F300
G1 X30.15 Y30.15 Z-0.25 F300
G1 X86.45 Y30.15 Z-0.25 F300
G1 X86.45 Y38.95 Z-0.25 F300
G0 X86.45 Y38.95 Z75
G0 X87.65 Y40.15 Z75
G0 X87.65 Y40.15 Z5
G1 X87.65 Y40.15 Z-0.25 F15
G1 X28.95 Y40.15 Z-0.25 F300
G1 X28.95 Y28.95 Z-0.25 F300
G1 X87.65 Y28.95 Z-0.25 F300
G1 X87.65 Y40.15 Z-0.25 F300
G0 X87.65 Y40.15 Z75
G0 X88.85 Y41.35 Z75
G0 X88.85 Y41.35 Z5
G1 X88.85 Y41.35 Z-0.25 F15
G1 X27.75 Y41.35 Z-0.25 F300
G1 X27.75 Y27.75 Z-0.25 F300
G1 X88.85 Y27.75 Z-0.25 F300
G1 X88.85 Y41.35 Z-0.25 F300
G0 X88.85 Y41.35 Z75
G0 X90.05 Y42.55 Z75
G0 X90.05 Y42.55 Z5
G1 X90.05 Y42.55 Z-0.25 F15
G1 X26.55 Y42.55 Z-0.25 F300
G1 X26.55 Y26.55 Z-0.25 F300
G1 X90.05 Y26.55 Z-0.25 F300
G1 X90.05 Y42.55 Z-0.25 F300
G0 X90.05 Y42.55 Z75
G0 X32.78 Y33.526 Z75
G0 X32.78 Y33.526 Z5
G1 X32.78 Y33.526 Z-0.25 F15
G1 X32.648 Y33.798 Z-0.25 F300
G1 X32.948 Y35.821 Z-0.25 F300
G1 X34.793 Y36.55 Z-0.25 F300
G1 X32.55 Y36.55 Z-0.25 F300
G1 X32.55 Y32.55 Z-0.25 F300
G1 X34.474 Y32.55 Z-0.25 F300
G1 X32.78 Y33.526 Z-0.25 F300
G90 G54 G64 G50 G17 G94
G49 G40 G80
G21
G30
M5
M30
```

voneiden, 2023-05-17:
If that's a 2d contour pocket using clipper then there should be a transition cut from one contour to another as well instead of a full retract like here

[image]

 Can't triage based on this though whether it's from your changes or whether the current implementation is just unreliable. Looks like there is NO test for this feature though, my bad.

giannissc, 2023-05-18:
Yeah I have been playing with the code yesterday and removed various functions from router.py to see their effect and I finally understood what shift_edges() does. Wouldn't a better strategy be to always transition on wire corners (or the start point to be more accurate since arcs and splines won't have corners)?

So this is the previous design but with the bug I introduced  fixed:
[image]

This is a second one that uses the shift_edges() function:
[image]

In my head this seem like a better presentation:
[image]

[image]

Am I missing something? @voneiden

voneiden, 2023-05-18:
> The closest_point and shift_edges() logic seem to be adding additional distance that the tool must cover. 

Don't quite follow that one, closest point should be the shortest path, no?

Anyway, my assumption has been that horizontal ramping is going to require being able to start the next contour/path from an arbitrary point so from that perspective the current implementation serves as preliminary work.